Remillard/VHDL-Mode #23: Investigate auto indenting settings additions

Remillard, 2017-08-02:
It would be nice to get automatic indent after some structures like "then" and so forth.  Typing in this mode and having to hit tab a lot has a strange feel to it, so it's something Emacs vhdl-mode was likely doing automatically.

Remillard, 2017-08-04:
Yeah this feature is going to take a lot of experimentation to make it work in a way that feels seamless.

emanuelen5, 2020-04-16:
This feature can be implemented using a TextMate setting file. Here's an example: https://github.com/SublimeText-Markdown/MarkdownEditing/blob/master/Indent%253A%20Raw.tmPreferences

Remillard, 2020-04-16:
Well, I know there's a method.  The trouble was that nothing I tried felt natural using it.  I haven't looked at this feature in a long time so perhaps when I get some opportunity, I can take another crack at it.