Remillard/VHDL-Mode #29: For another editor

Alexey-T, 2017-08-30:
Hi, is it easy to port plugin to CudaText editor?

Remillard, 2017-09-07:
I honestly do not know.  The language behavior has been mostly compartmentalized against the Sublime Text API but I would not swear that it is 100% isolated.  I do not know anything about CudaText and extending this to that editor is currently beyond the scope of my time and efforts.

Alexey-T, 2017-09-08:
I can port the plugin .. But I need isolated py modules from ST, and others must use them, can you do it?

Remillard, 2017-09-08:
Like I said, I know nothing about CudaText, so I don't know if I could.  However like I said, I've tried to isolate behavior from Sublime Text API.  If you'd like to take a look at it and do the port, you are more than welcome to.  This is the general structure.

`vhdl_mode.py` : This module is primarily associated with Sublime Text API and the command set.  
`vhdl_util.py` : This is also somewhat Sublime Text API associated, mainly short utility methods that abstract some API call to something more useful for my purposes.
`vhdl_lang.py` : This is the module that knows VHDL and it does not rely on the Sublime Text API directly but it does rely on the calling methods doing things properly.  For instance, I did write a command line beautifier that uses the methods in `vhdl_lang.py` which was quite handy for debugging.  However it has no access to some useful things I can get from the API like the scope of lines, so some of the routines don't work quite as well.  

Good luck!

Remillard, 2017-09-18:
I hope this helped you out.  I haven't heard anything for over two weeks, so I'm going to assume you got the information you wanted or needed regarding porting to another editor.  I wish you well.  If you reuse my work I'd appreciate attribution as it says in the licensing file.  I'm going to close this issue for now.  I think it can be reopened if need be.

closed 2017-09-18 by Remillard

Alexey-T, 2017-09-18:
Sorry that i didnt write: it is hard for me to make port...