Remillard/VHDL-Mode #30: Stutter Typing, incompatible with emacs vhdl mode

Quenii, 2017-09-07:
ie
'[' for '('   
'[[' for '['
';;' for ':'

Remillard, 2017-09-07:
Ahh, the stutter typing may be slightly different then.  I never used the square bracket for paren replacement.  Also I did use `;;` for `:=` because I found that more useful.  Colon is relatively easy to type.  What we really like to replace is the `=` mark which is used in every assignment (left arrow, right arrow, and variable).  A single colon is already above `;`.  

So I may simply put in a section of the documentation indicating that I've made a break here.  However!  It would be really easy to add this to your own Sublime Text behavior if you wanted.  The solution to stutter typing was to put in a keybind that inserted a character.  If you look at the `Default.sublime-keymap` file you should see a section where I put in the stutter typing command.  This is one example:

```
	// Stutter typing keybinds.
	{
		"keys": [".", "."],
		"command": "insert",
		"args": {"characters": "=> "},
		"context": [{"key": "selector", "operand": "source.vhdl"}]
	},
```

You can override this file pretty easily and add in your own stutter typing.  And I may put in the `[` and `[[` ones but I'll have to play with it a bit and make sure it feels correct to do it that way.  I'll also back check against emacs for this.

Thanks for pointing this out!

Remillard, 2017-09-07:
I'm looking at the Emacs stuttering and I think I can implement these more properly.  The only issue is two `;` versus three `;`.  The way Sublime Text's keycapture system works is that if a key sequence exists for something where there is ambiguity, it puts in a potentially annoying delay.  For example I did note in the documentation that I altered the comment command behavior because doing `---` for a line and `----` for a box caused some awkward rhythm.  So I am inclined to keep `;;` for `:=` as an incompatibility.   But I can definitely alter the `[` and `]` behaviors as I noted before and I don't see any real problem.  

Also, it looks like I don't put in an extra space at the beginning which is an error.

I'll get this altered hopefully in the next few days.

Remillard, 2017-09-07:
Documentation note to self (in case I'm working on this on a different machine than usual) 
```
  STUTTERING:
    Double striking of some keys inserts cumbersome VHDL syntax elements.
    Stuttering can be disabled (enabled) by typing `C-c C-m C-s' or by
    option `vhdl-stutter-mode'.  Enabled stuttering is indicated by `/s' in
    the modeline.  The stuttering keys and their effects are:

      ;;   -->  " : "       [   -->  (        --    -->  comment
      ;;;  -->  " := "      [[  -->  [        --CR  -->  comment-out code
      ..   -->  " => "      ]   -->  )        ---   -->  horizontal line
      ,,   -->  " <= "      ]]  -->  ]        ----  -->  display comment
      ==   -->  " == "      ''  -->  \"

```

Remillard, 2017-09-07:
Yep, I was afraid of that.  `;;` and `;;;` do NOT work together well.  Since there is ambiguity Sublime Text will wait for the 3rd keystroke to render.  This does not bode well for the doubled up ones like `[[` also.  I should just put in a caveat in the documentation as I don't think I can properly duplicate this behavior.

I was able to make sure the extra spaces for the other commands are rendered correctly per vhdl-mode though.  I'm going to check on the ST Discord channel to see if anyone has a suggestion on how to tweak the keybind engine.

Remillard, 2017-09-07:
There might be a way to handle this with key binding contexts but it's gonna get ugly.

Quenii, 2017-09-09:
I think ";;" for ":" is to avoid using shift, '[' for '(' the same
to solve the ";;" ";;;" ambiguity, you could type ':' by ';;', and than type '='

Remillard, 2017-09-09:
Well I understand why it's done, just that the ambiguity does not resolve well in Sublime Text in a way that feels natural if you're used to Emacs.  I'm still playing with it.  It's something that might be able to be accomplished in a command triggered by a keystroke rather than entirely in bindings.  And ultimately, it may not be able to be resolved at all.

Remillard, 2017-09-11:
Found something that works pretty well.  I have no access to the keystroke stream, so I cannot emulate this 100% perfectly, however if I create a keybind for the shorter (ambiguous) replacement, then do a context check on a single keybind of the same character, and if it's the replaced text, then run a macro that removes and re-inserts the text.  This won't work well for the comment box.  Checking for an ambiguous row of comment dashes will be wrong if someone just holds down the - key.  However I think I can create the simpler replacements this way where there is ambiguity.

Remillard, 2017-09-11:
Actually, I'm going to leave the square brackets and parenthesis alone.  If I do it with this method, it would be impossible to do nested parens because the keyboard bind would search for preceding ( and replace it with [.  And again, I don't have direct access to the keystream, so I can't do a custom timeout.

Remillard, 2017-09-11:
Well I think I've done what I can here.  Really the only thing I could handle well was `;;` versus `;;;` but I did that, and then documented why I cannot handle the other stutter constructs.  I've got this change staged for the next release, and will close the issue.

closed 2017-09-11 by Remillard