Remillard/VHDL-Mode #87: Package Control: Install Package does not install the plugin

Artavazd-B-Goode, 2018-02-24:
It seems that there is an error when trying to install vhdl-mode package trough Package Control for ST V3, Build 3143. The process of installation stops after a short period of time. Tried to run ST with Admin privileges but faced with the same problem.

Remillard, 2018-02-27:
I haven't had a chance to check this, however I went to the Package Control website.  I checked the VHDL Mode page and there are three days of no installations, however it picked back up today.  I think there was something out of whack between Package Control and the GitHub API.  Since it's back and going today, I'd suggest that you try installing it again.

Artavazd-B-Goode, 2018-02-28:
Hi Remillard, 
Thanks for reply. I have checked one more time and package installed correctly. Smth was wrong with website.

Remillard, 2018-02-28:
Glad you got it working and I hope the package is a benefit to your coding work.  Cheers!

closed 2018-02-28 by Remillard