Remillard/VHDL-Mode #89: Function beautification still has issues

Remillard, 2018-03-09:
The function beautification is not returning back to normal if the return type has a vector index
```
	-- Padding function
	function make_miso_word (
			d_in : std_logic_vector) 
		return std_logic_vector(7 downto 0);
		
		end package csm_pkg;	
```
However if that (7 downto 0) is removed, the indent on the following lines is correct.

Remillard, 2018-03-09:
Some more trouble with function beautification
```
	-- Padding function	
	function make_miso_word (d_in : std_logic_vector)
		return std_logic_vector is
		variable d_out : std_logic_vector(C_SPI_DATA_LEN-1 downto 0);
	begin
	end function make_miso_word;
```

If this is beautified, the 'return' line will be aligned with the starting line.

kalund, 2018-08-31:
Great plugin first of all. I have a similar issue with procedures with multiline input parameters. Example below:

    procedure test (
        constant test : string
            ) is
    begin
        log(test);
        end procedure;

Remillard, 2018-08-31:
Thanks for letting me know.  The subprogram work was the last thing I added to beautification and it still has some issues.  I'm still trying to find time to work on this but have been swamped with firefighting work projects.  I'll try to prioritize this.

Remillard, 2019-05-16:
Sorry that this took so long to sort out.  It was actually a multiple issue problem.  First there was some issue with the closing parenthesis on subprograms.  Secondly, I needed a way to ignore the constant.  See because you have constant on its own line with no semicolon, it tries to follow the constant rule which in any sort of declarative region requires a semicolon to close.  In an interface list, constant does NOT require this, which breaks the rule.  Over winter I did add functionality to make beautification ignore some cases.  Adding subprograms to that seemed to help along with the closing rule issue.  Going to be testing this a bit and hopefully getting it out.

closed 2019-05-20 by Remillard