Remillard/VHDL-Mode #93: VHDL configuration syntax completion

VincentSmit, 2018-03-15:
First of all, amazing package. Made my vhdl-life a lot easier. I had one feature request for this package. Could the [configuration](http://www.vhdl.renerta.com/mobile/source/vhd00020.htm) syntax be added to the syntax completion? Configurations are a great feature within vhdl when it comes to setting up testing swiftly. 

I have the feeling that it would not be too hard, and I am willing to write the code for it. It would be my first time writing/adding to a sublime package though, and I do not want to pollute your code with a possible messy addition. Therefore I just opened an issue rather than creating a pull request directly.

Remillard, 2018-03-15:
You're looking for a snippet for configurations?  That wouldn't be too difficult to add, though the variations on configurations, if I remember correctly, is fairly large.  The only issue with snippets is that they don't do branching syntax very well.  However I'll look into it!

Very glad this is working for you.  Hopefully I'll get a chance to bang on it some more.  I found some things about function beautification that I didn't like that I need to also work on.

If you come up with a snippet that does what you're looking for, I am quite willing to add it to the package in general after some review.

VincentSmit, 2018-03-27:
Thanks for your quick response. I did not have time for attempting this yet, but I'll give it a go this weekend. I'll get back to you if I get something reasonable.

Remillard, 2018-03-27:
Sounds good.  Unfortunately I've had zero time to work on this or the other issues lately.  Maybe I'll get some freeform slack time soon, but it's difficult to say.

Remillard, 2019-01-03:
Sorry it took awhile, but there is an upcoming snippet for this feature as soon as I can test 1.8.0 enough.

Remillard, 2019-01-03:
Of course this uncovered a bug in beautification!  Apparently my rules for configurations are not sufficient for indenting them properly.

Remillard, 2019-01-04:
New rule for `for` that is under loop and generate, so it ought to get caught last out of the many uses of `for`.

closed 2019-01-04 by Remillard