Remillard/VHDL-Mode #94: Visual Studio Code

DRuiz42, 2018-06-21:
Any chance of seeing a version for Visual Studio Code?

Cheers

Remillard, 2018-06-21:
I don't use VS Code.  I understand it outsources language handling to a sort of Language Server module.  I looked into VHDL variation on language servers but most work there seemed pretty early.

Anyway, if someone wanted to collaborate who understood the LSP framework, I'm open to it.  There are definitely some challenges in handling syntax and beautification.  However for my own part, Sublime Text does what I need and want it to, so porting to VS Code by myself is unlikely.

DRuiz42, 2018-06-22:
Thank you for your answer. I was asking that because I am afraid the ST community won't survive once it will require a license. I will try to take some time to look into the LSP framework.

Remillard, 2018-06-22:
Sublime Text is already a licensed product and has been doing fine for years.  I don't think there's a great deal to worry about here.  In any event, the plugin is in Python, easily readable and probably easily portable.  The only aspects that are Sublime specific is the commanding interface and I do use the lexical scoping engine in places.

closed 2018-07-13 by Remillard