Remillard/VHDL-Mode #104: VHDL syntax validation

lenigoor, 2018-08-27:
First I would like to express my thanks for making this plugin. While writing some VHDL code I found that the following snippet is incorrectly highlighted. Maybe this can be fixed?

[image]

Sv3n, 2018-08-27:
The match rule on https://github.com/Remillard/VHDL-Mode/blob/master/Syntax/VHDL.sublime-syntax#L1944 is too broad: it eats the opening bracket as the non-word character before the tick char, preventing the basic-paren-group from matching properly:

```yaml
  character-literal:
    # Any of the printable characters in the standard character set can be
    # used.  Used the non-word character element to protect against
    # single quote used for attributes (e.g. a'range)
    - match: '[^a-zA-Z0-9]('')'
```

A possible fix:

```yaml
  character-literal:
    # Any of the printable characters in the standard character set can be
    # used.  Used the non-word character element to protect against
    # single quote used for attributes (e.g. a'range)
    - match: '[^a-zA-Z0-9\(\)]('')'
```

(I was staring at a similar issue coincidentally)

Remillard, 2018-08-27:
Thanks!  I'll get that staged.

Remillard, 2018-08-27:
In the process of making changes, I started experimenting a bit.  Looks like it's wholly a problem with the single quoted literal.  Note on adder2 the first parenthetical group works fine.  If you add/subtract spaces on adder1 you can make it pass/fail.  Indeed the character literal is catching the opening paren there though.  I think it's the fact that I have `('')` in there but trying to remember for the life of me why I put that in to begin with as I don't think a character literal can start that way.  Still investigating.

Remillard, 2018-08-27:
Oh wait, I'm trying to capture the opening `'` mark.  However I think the parens are confusing things.  I think it's just a regex issue.

Remillard, 2018-08-27:
Okay, after discussing this with Thom (kind of a syntax engine guru) on the Discord channel, I think I've pieced together what probably needs to be done.  It's actually kind of subtle so I'll explain.

So, identifiers in VHDL can be almost anything and they tend to match everything, so I have them at a very low priority in matching contexts.  Additionally, "attribute modified identifiers" are not captured atomically.  Basically a word falls through and gets marked `variable.other.vhdl` and then the attribute falls through and gets marked `keyword.other.vhdl`.  The attribute punctuation doesn't get scoped.  

However, this means that there's a free punctuation that has escaped!  The attribute punctuation is sort of a free radical and then can match character literal.  So to fix that problem I put in the negation group which then unfortunately captured lenigoor's and Sven's opening parenthesis as well.  So my fix to the problem of attributes caused this problem.

So... I think (unfortunately) the solution is to fix attribute modified identifiers properly.  I cannot just let them fall through to identifier and keyword, but I have to identify them as an atomic whole.

Long story short, this is going to be a tricky fix.  I think I know what needs to be done (create a special context group for this) however there's a lot of testing to do to make sure it doesn't break anything else.

Remillard, 2019-06-17:
This actually ended up being a relatively simple fix actually.  Looking backwards cannot be done well with the syntax engine for speed.  It is actually possible but Sublime switched to a much slower regular expression engine when you have look-behinds.  However I'm trying to find single character literals which is a pretty simple construct.  Instead of using look-behinds, a look-forward for a single character and quote mark seems to do the trick.  This does not consume the `'` for either attributes or for typecasting and does not rely on the preceding character not being a alphanumeric.

Remillard, 2019-06-17:
[image]

Since some screenshots had been shared before, thought I'd throw this in there.  I'm copying @lenigoor 's original code.  The parentheses are captured, and should be noted that the single character capture is space agnostic.  Additionally, the attribute tick is honored, as is the string cast.

Remillard, 2019-09-06:
There exists a case with a typecast bit that fools this.  `std_logic'('0')`

commit edd0cd40 (2019-06-17): Merge pull request #137 from Remillard/develop

Fixed #104 Finally -- Woohoo!
closed 2019-11-27 by Remillard