Remillard/VHDL-Mode #119: Paste ports as components

larshb, 2019-02-04:
[image]

Pasting ports as component gives this result. Cropping off the last character in whatever type the last signal is. Removing the closing parenthesis if it is an array type.

Remillard, 2019-02-04:
I will check this out this morning.

Remillard, 2019-02-08:
Did this help or does the problem persist?

Remillard, 2019-02-11:
I believe this problem has been fixed in version 1.8.0 and since I've not received a reply, am going to close this issue.  If you continue to see this behavior please let me know.

closed 2019-02-11 by Remillard

larshb, 2019-02-19:
Thank you for your reply. I am no longer able to replicate the bug myself, so I guess 1.8.0 fixed it.

Remillard, 2019-02-19:
Excellent!  Glad it's working better for you.