Remillard/VHDL-Mode #122: Beautify adds trailing spaces

pkorpine, 2019-02-26:
Beautify adds a trailing space to lines that end to a semicolon.

Example:
- before beautify:
[image]
- after beautify - a trailing space is added:
[image]

Tested on v1.8.1.

Remillard, 2019-02-26:
Yeah this is a known issue.  I've classified this as a minor behavior, perhaps not even a bug as it doesn't actually affect anything other than a sort of invisible aesthetics.  

Sublime Text also has a setting for this that solves the problem.  If you set:
```
"trim_trailing_white_space_on_save": true
```
in your personal settings, it'll remove this space.  

Hope that helps the issue in the meantime.

pkorpine, 2019-02-27:
Thanks for the quick response and also for the excellent plugin!

I'd like to keep this as a (minor) bug as it adds white-space problems in version control.

My workaround is to redefine the beautify key binding with a macro that runs `vhdl_mode_beautify_buffer` + `trimmer`.

Remillard, 2019-02-27:
Yeah I could probably go and write a method that does another pass at the end of beautification to remove trailing whitespace.  However, how does this affect version control as the file would have to be saved before VC will pick up changes and Sublime Text will also remove whitespace at the end of the line?  Granted, it requires using that setting so if someone wasn't, then VC would pick up the change.

Remillard, 2019-03-13:
I wondered how you discovered this because it's pretty esoteric and you'd need to have saved the file before an external git client would discover that the line endings had changed.

Then I got Sublime 3.2!

With the diff bar in the gutter after beautify, the entire bar turns different, but if you immediately save it'll basically change to just the changes you made.  So that's visible and super obvious.  That does push the need for a fix a lot faster than I previously thought!

closed 2019-03-27 by Remillard