Remillard/VHDL-Mode #128: Operand overload

ni-maxkoehler, 2019-04-24:
Syntax highlighting for operand overload is not working.
[image]

Remillard, 2019-04-24:
Thanks for letting me know!  I'll look into it.

Remillard, 2019-05-16:
Believe I've got a fixed staged for this now.  Should be released fairly soon as soon as I test some other things in the release.

ni-maxkoehler, 2019-05-17:
Thanks for taking care about this.
I had a quick look into the change. Does the regular expression captures function names like "and" as well?

Remillard, 2019-05-17:
Mmm, probably not if overloading the logical operators requires the use of double quotes.  I looked through the abbreviated LRM in the Ashenden book, and then glanced at the section for operator overloading.  It was somewhat obscured if overloading those operators required quotes and I figured if it doesn't then the usual valid identifier match would work.  Easy enough to add though so I can do that.  I'm off work for retrieving kid from college today so don't know if I'll get to it today, maybe over the weekend.

Remillard, 2019-05-20:
And indeed the overloading for keyword operators does also require double quote marks so I'll have to add that to that list.  In addition, the end of the body is not scoping right so I"m not quite there.

closed 2019-05-20 by Remillard