Remillard/VHDL-Mode #133: "block" syntax not recognized by syntax highlighting or beautification

rolpks, 2019-06-14:
I use `block` often. E.g.:
```
  SomeBlock : block is
    signal x : std_logic := '0'
  begin
    x <= '0';
  end block SomeBlock;
```

But although VHDL mode has a snippet to create a block, this syntax doesn't seem to be supported. Syntax highlighting looks incorrect:
[image]

Compare that with how an if-generate looks:
[image]

I would've expected `block`, `begin`, `end` and `block` to be purple here. `variable` should be purple, too, looks like its syntax highlighting is being thrown off by the unrecognized keywords before.

Aditionally, alignment is broken for this keyword, too. After beautification, I get this:

[image]

Remillard, 2019-06-14:
Hey thanks for letting me know.  I can honestly say I've never used this construct, and thought I thought I was being thorough in the syntax definition I clearly missed this one.  I'll see if I can't get that added to the syntax list and then figure out where it falls in the realm of beautification.

Remillard, 2019-06-14:
Out of pure professional curiosity, what do you use this construct for in a practical sense?  The only thing I can come up with is when you need a local signal (say for a counter), don't want to use a variable for some reason, and don't want to put it into the architecture's declarations.  However I'm very curious if there's another reason to use it.

Remillard, 2019-06-14:
Alright, wow.  I can see why I might have skipped this one.  It could be the most lexically complicated concurrent statement there is.  Not only does it have its own declaration and statement blocks, but also allows for a logical expression to activate it AND generic and regular ports AND generic and port maps also.  And then it permits any other concurrent statement inside itself, presumably including itself.

Well you're using it, so I'll see what I can do!

rolpks, 2019-06-14:
> The only thing I can come up with is when you need a local signal (say for a counter), don't want to use a variable for some reason, and don't want to put it into the architecture's declarations.

This is mostly how I use it. It gives locality to declarations (signals and constants) that are only relevant to a small subset of the code, and makes it so the declarations live right next to where they're being used. It makes code more readable, so it's been adopted as part of our design guidelines.

> Not only does it have its own declaration and statement blocks, but also allows for a logical expression to activate it AND generic and regular ports AND generic and port maps also. 

I've certainly never used it to _that_ extent. The "guard expression" I've never used because I don't believe it's supported by most synthesizers. I didn't realize you could have ports in there, but it certainly acts as a mini-architecture, and you can put anything in there that can go in an architecture (both in the declaration and statement section). Including, like you mention, other blocks.

Remillard, 2019-06-14:
I can see that.  It's interesting that in its simpler forms it is actually synthesizeable.  

Here's the actual definition of this beast in the LRM (well the version of it in Ashenden's book).  Anything in `[]` is optional.  
```
block_statement <=
    block [ ( guard_expression ) ] [ is ]
        [ generic ( generic_interface_list ) ;
        [ generic map ( generic_association_list ) ; ] ]
        [ port ( port_interface_list ) ;
        [ port map ( port_association_list ) ; ] ]
        { block_declarative_items }
    begin
        { concurrent_statements }
    end block [ block_label ] ;
```

And writing that I JUST realized that the keyword `is` is OPTIONAL.  I had this working pretty nicely so far for syntax.  I'm not sure I can actually support the optional nature of is and still write syntax for this.  The way Sublime's syntax engine works is that it allows me to define "zones" (for lack of a better term) where only certain things apply.  This is why it ought to not highlight correctly if you try to put an entity inside an architecture body.  It's illegal.  However with that guard expression being a logical operator, I put the logical operator zone there, and then marked that off with `is`.

Ahh well, maybe I'll just make it work the way it's working and if someone complains to me about the optionality of `is` I'll point them here!

Interesting use for the design guidelines.  I can see that as a potential benefit.  I think in most cases where I really want something local, I'll use a variable, though naturally this wouldn't work for concurrent statements.  And if I were maintaining code, I'd not want to have to parse through the body of the code for constant declarations.  I feel those should be at the top or in a package.  However there's about a dozen billion ways to skin a cat in VHDL and I can see what you're doing. 

In any event, the syntax scoping is working out alright thus far.  Beautification is an entirely different system so I'll have to go see how that might work.

Remillard, 2019-06-14:
Okay, closer now.  Seems to scope and beautify correctly.
[image]

Once I'm satisfied with coming up with test cases for this construct, I may need you to do testing for me because I do not have a lot of "live" examples of how this thing works.  Current issues are that beautification if it uses generic/port mappings gets blown up, but otherwise things look fairly pretty.

rolpks, 2019-06-14:
Looks very promising. I'll be happy to try this out on a bunch of files I have lying around.

Remillard, 2019-06-14:
I fixed up a few other things I noticed along the way and put out a release.  You should see it come up once Package Control picks up the modified release.  I'm going to go ahead and close this issue for now BUT if you find something broken I'm quite happy to reopen it.  (Well not happy per se, because I don't like things being broken, but happy to have a place to discuss the issues.)

closed 2019-06-14 by Remillard

rolpks, 2019-06-17:
Seems to be working alright. It did make me realize that I often *don't* use the `is` keyword, but that's easily remedied. Thanks!

Remillard, 2019-06-17:
Sure thing.  Thanks for letting me know.  I did modify the block snippet a little so perhaps it'll make it somewhat easier to keep `is` included.  If I find a way to distinguish that guard expression zone from the other stuff without needing `is` as a barrier, I'll fix that, but for now I'm afraid that's the way it has to be, at least if I want to meaningfully scope everything else.  Syntax scoping is pretty much an extreme thing.  Either you just highlight keywords and not much else, or you almost have to attack it as a tokenizer and catch EVERYTHING.