Remillard/VHDL-Mode #136: Add GitHub issue and pull request templates

dominiksalvet, 2019-06-14:
It would be great to add some issue templates and a pull request template to follow the GitHub workflow. If you want, I will prepare some simple ones for you that can be further modified and then I will create a pull request. Shall I?

Remillard, 2019-06-14:
I appreciate the offer however I don't think that this project is big enough to worry about such things.  With one rare exception I'm the only one who ever PR's to the project.  And for issues, I'm happy enough that someone bothers to report a bug, let alone worry overmuch about the format of their request.  If ever there became enough users and developers to require that level of management, I'll check into it then.

dominiksalvet, 2019-06-14:
@Remillard Well, I didn't intend to do it because of management. I personally find contributing to projects with templates more comfortable because I know exactly where I should put each piece of information. But it looks like it is only my subjective feeling.

Nevertheless, thank you for your work. I really :heart: it. Just press a key and boom -- my VHDL code is beautified. :)))

Remillard, 2019-06-15:
Well, it's pretty casual around here, so you should just relax :)

Glad you are getting value out of it.  Hoping to refactor it to something slightly more sane, as soon as I can figure out how to do the same thing I do in another method.

closed 2019-06-17 by Remillard