Remillard/VHDL-Mode #141: Beautification Indent of Nested WHEN / IF statements

jozwa, 2019-08-18:
how i hoped it would look:
[image]
[good.txt](https://github.com/Remillard/VHDL-Mode/files/3513079/good.txt)
how this nice plugin Beautifies it
[image]
[bad.txt](https://github.com/Remillard/VHDL-Mode/files/3513080/bad.txt)

Remillard, 2019-08-18:
I think that's happening because you're putting all those statements on the same line.  The beautifier is not a complete word-by-word tokenizer.  It works on finding keywords on lines and largely as soon as it finds a keyword on the line, it'll decide what to indent.  If you put too many things on the same line, it'll have problems and there's no way around that.  Notice on the `when s3` clause it indents properly.

I'm not a big fan of "everything on one line" but if you must, try not putting it on the same line as the `when` and I think you'll get slightly better results.  I'm honestly rather surprised it doesn't freak out with a full `if/else/end if` on the same line.

Remillard, 2019-09-04:
Out of curiosity, I tried rearranging the beautification rules to see if I could get a combination that would support this code style, with hilariously bad results.  

Then I got to looking at it more seriously as to why it was behaving the manner shown when I used a combination that I thought had a chance of working and it boiled down to the fact that I used a general case end clause for most expressions (in VHDL usually it's just `end <word>`  and if I made specific ending clauses then it might distinguish better.  And indeed, this worked out and I actually got something like your code to beautify.

I am going to keep this one under observation for awhile though because there are a lot of hard to predict effects.  However assuming nothing bad happens, I think I can support your code style (though I still think it is not a good style).

jozwa, 2019-09-05:
Thank you for you reply. I did forgot to reply.
Now you say, i did make a mistake in the good example. I will add a new one at the bottom.
maybe it is possible to check on the first keyword to define the indent on that line and on the if / end if on the same line, again up and down, so no change at all.

[image]

Remillard, 2019-09-05:
It's somewhat more complicated than that, however yes I approach it from a line by line perspective.  I scan for keywords in an order and when it finds one, that is typically (not always) sufficient to define the indent for that line.  I thought initially that making `case` and case's `when` (which is different from other uses of `when`) a slightly higher priority than if might solve the problem but it did not.  What was happening was that it was triggering the end of the case structure on the unconsumed `end if` because I have a very general closure statement that works for most things.  The `end if` is unconsumed because it never saw the `if` because it's on the same line as the `with`.

The proposed solution was to keep that priority, but create more specialized closure clause for `case` and `if`.  This has seemed to work okay in the test cases I've thrown at it so far, but I'm still evaluating.  I may have to do some if/generate cases and see if I broke anything there because that CHANGES the end phrase (like I said, the generic worked pretty well for a lot of things).

Remillard, 2019-09-05:
Shockingly, if/generate seems to work out acceptably.  I might have to look up case/generate.  I tend not to do a lot of generate things in professional life.

Remillard, 2019-11-25:
Haven't seen any side effects yet, so will call this one handled for now and should be in next release.

closed 2019-11-25 by Remillard