Remillard/VHDL-Mode #147: Syntax highlighting

lenigoor, 2019-11-28:
The following construction is allowed with the "new" VHDL-2008 syntax, but the highlighting on the last bracket does not agree.

```vhdl
architecture rtl of test is
	signal alpha : std_logic_vector(1 downto 0);
	signal beta  : std_logic_vector(1 downto 0);
	signal gamma : std_logic_vector(3 downto 0);
begin
	(alpha, beta) <= gamma;
end architecture;
```
[image]

Remillard, 2019-11-28:
That’s a 2008 construct? I am not surprised that I don’t support that. I’ll have to research that and see if there’s anything I can do. The assignment construction is one of the most problematic because it is so generic.

It’s Thanksgiving in the US so will probably be a few days before I can dig into this. Thanks for the report. I really appreciate it.

Remillard, 2019-12-02:
Alright.  I can't say I see this structure a lot but maybe I should.  It seems useful.  In any event I found notes in Ashenden (page 102 in 3rd edition) on aggregate assignments in VHDL 87/93/2002 (where it was much less useful) and then in 2008, it does seem to allow a lot of possibilities.  I'll have to see if that's something I can work into the signal assignment context rules.  It'll probably take a special match.

Remillard, 2019-12-03:
Basic support added.  May be able to build on that in the future.

lenigoor, 2019-12-15:
Awesome, thank you for adding some basic support 😄

Remillard, 2019-12-15:
No trouble at all.  If I can figure out how to scope it better, I'll do so though I'm worried it would require a complete restructuring of the syntax definition.  And I have a sneaking suspicion it might break if you try to aggregate assign to slices, but it's a little step forward.  Honestly I probably ought to start using this construct in my own code; it seems pretty handy.

erpalma, 2020-01-16:
Can you please also add the variable assignment operator (:=) to this fix? It's VHDL 2008 too.

Remillard, 2020-01-16:
Sure I'll see what I can do.