Remillard/VHDL-Mode #162: VHDL-2008 syntax highlighting not working

CrunchyNougat, 2020-08-03:
Hi Remi,

just recognized that the syntax highlighting and color encoding is confused when for example "case?" is used in the code.
I guess that depends on the vhdl-mode package?

Remillard, 2020-08-03:
I believe you'd have to show an example.  I'm not sure if you literally mean `case?` which is not a VHDL construct.  If it's a brand new 2019 construct I can pretty much guarantee there's no support because I got started on working through that at the beginning of the year but haven't had an opportunity to do much with it since springtime.  `case` is a construct and it does highlight and style properly.

CrunchyNougat, 2020-08-03:
Hi Remi, as far as I understand it is a vhdl-2008 feature:

https://s3.amazonaws.com/verificationhorizons.verificationacademy.com/volume-8_issue-3/articles/stream/vhdl-2008-why-it-matters_vh-v8-i3.pdf

And in my code it looks like this, when using the case? or not:
[image]
[image]

Remillard, 2020-08-03:
Well I went back to good old Ashenden and lo and behold in the LRM summary appendix, it does list ? as an optional character and going back to a secondary discussion on case in the text (page 125 to be specific), it explains that it's used with don't care's in std_ulogic.

While potentially a handy construct, I can guarantee you it won't match nor will it probably beautify correctly.  I apologize for the oversight and I'll add it to the TODO list.  Unfortunately I can't tell you when it might be fixed as I've had no time to really work on VHDL Mode for many months.  This is kind of a glaring omission so I'll try to do what I can when I can spare the time.

CrunchyNougat, 2020-08-04:
Hi Remi,
thank you for the answer!
I also recognized it is a rather less used feature in vhdl. Nevertheless, I started with verilog, and here the casex construct is widely used since it makes possible very lucid state machines.

However, I think it is just the keyword "case?" missing in the list of protected language terms. Is it possible for me to add it to my package easily? Thanks!

Remillard, 2020-08-04:
Doubtful.  If you are curious as to how we define syntax in Sublime, take a look at the file Syntax/VHDL.sublime-syntax.  It's not just matching keywords.  We define entire lexical structures so that the entire structure is identified, including all internal parts.  It MIGHT be as simple as adding an optional ? character, but I wouldn't swear that there aren't any knock-on effects. 

On a general VHDL note though, you would not ever use `case?` for state machines in VHDL.  It has a relatively singular purpose that derives from the fact that VHDL is a strongly typed language and the common type used is `std_ulogic`.  `std_ulogic` contains the don't care symbol `-` however it's treated as any other symbol and thus something like the conditional `"10--" = "1010"` would fail because `-` is not `0` or `1`.  Thus the function `std_match` was created.  So if you needed to do a lot of don't care matching it required a `if` / `elsif` structure to go through a lot of matches. 
 
This was deemed a little tedious and I guess they invented `case?` in 2008 to get around the use of `if` and `std_match`.  

In a state machine, I would highly recommend using enumerated_types.  That's one of the things they are there for.  Long sections of don't care matching are largely the domain of address matching and honestly I'd just use `std_match` (especially since I was unaware of `case?` until you brought it up.

That said, it's not an argument for excluding it from VHDL Mode.  I'll try to get it covered when I can.

CrunchyNougat, 2020-08-05:
Thank you for your detailed answer!
I think it goes a bit too far for an issue description but just to clearify:
Of course VHDL is a strongly typed language and I know about the fact using types for state machines. But, the type is used to define the states of a state machine, the way I am using the case? is to describe the transition logic. And here I gain a lot of transparency when having all transitions presented in a simple and easy way to verify. 
First: I can directly check for exclusivity.
Second: Even if the number of states increases the transition logic stays manageable.

Remillard, 2020-08-05:
Well you don't work for me, so good luck! :D 

In any event, I'll try to get the 2008 construct working when I can devote some time to it.

CrunchyNougat, 2021-01-08:
Hi Remi, still enjoy the use of your plugin very much. I am just wondering if you might have had some time to spend on the vhdl-2008 syntax highlighting and code beautification.

Thank you and best wishes!

Remillard, 2021-01-08:
Actually I"m getting a little bit of a project transition so there might be some time before anyone notices :D.  I don't know if any of the big stuff will get done, but might not be too time consuming to add the case?.  Suspect it'll just be a little optional character for regex's in syntax and the beauty rules, but I just haven't dove back in in awhile.

Glad you're still finding it useful.  I pretty much am in it daily myself.  I still wonder about better ways to do things.

closed 2021-03-03 by Remillard