Remillard/VHDL-Mode #167: VHDL beautification

JYH25, 2021-11-04:
Sorry to bother. I am confused about which package I should download to make VHDL code highlight like yours.

[image]

Remillard, 2021-11-04:
Well... I made that a LONG time ago.  I have no actual memory of what color package I was using for that.  However I went to my current list of color schemes and just used the up/down arrow and moved through them.  I *think* that this is probably "Monokai"
[image]
It has the italicized port names, the red mode, and the italicized type indicator in cyan.  

I don't currently use Monokai though as it's a little too brown for my taste.  Some current favored scheme bundles:
* Agila (has lot of variations, including a Monokai with extended capability and others)
* ayu (only 3 but covers dark, darkblue, and light).

I current use "ayu-mirage":
[image]

Hope that helps!

JYH25, 2021-11-04:
GET! Thanks a lot for your reply.

closed 2023-08-18 by Remillard