Remillard/VHDL-Mode #171: The settings file could not be opened when git cloning repo directly into Packages folder.

rhasst, 2023-08-18:
### Environment
    Ubuntu 22.04.2 LTS
    Sublime Text Build 4143

### Steps to Reproduce
-  `cd ~/.config/sublime-text/Packages`
- `git clone git@github.com:Remillard/VHDL-Mode.git`
- open sublime
- Preferences -> Package Settings -> VHDL mode -> Settings

### Current Behavior

The following popup appears `The settings file "res://Packages/VHDL mode/vhdl_mode.sublime-settings" could not be opened.`

### Expected Behavior

User settings should be opened.

### Solution

Problem is that `res://Packages/VHDL mode/vhdl_mode.sublime-settings` does not exist. Because package name is `VHDL-mode` and not `VHDL mode`. Solution is to rename `...Packages/VHDL-mode` to `Packages/VHDL mode`

Remillard, 2023-08-18:
Well as you described, the solution is that the default name that a git clone uses is not the name that Sublime Text uses.  Sublime Text expects the package name to have the space.

I recall running into this problem WAY back when I first developed the package and there was a kerfuffle between the repository name and the Sublime Text name.  I am not going to change the name of the repository in Github as that has some knock on effects with packagecontrol.io and the whole package ecosystem for Sublime Text that I do not wish to navigate at this time.  Additionally the expected way to install the package is through Package Control.  

I think the only thing I could do, and I will take this as an action is to add some documentation in the README that describes this step required if someone wants to clone the package directly without using Package Control.  However generally speaking if you are not going through the normal method of installation, you're largely on you own -- it's not supported.

But I can document that fact.  I've also got a pending snippet someone submitted that ought to roll into the mainline branch too.

rhasst, 2023-08-18:
I know that I am not installing it the standard way, I was interested on how it works and wanted to play around with it. Documenting it in README might be a good idea.

By the way, huge thanks for creating this package! It is awesome and I've been using it for quite some time. It is the only thing that is holding me back and not going to vscode.

Remillard, 2023-08-18:
I'm very glad it's been useful.  I have colleagues at work who have used it for years too at this point.  I'll admit, I don't do much to the package anymore, but it's withstood time fairly well.  I've kind of moved back to Emacs myself, but I still keep an eye on the package repository.

closed 2023-08-18 by Remillard